ZakKemble/AVRDUDESS #2: Bit clock (-B) drop down values are outputting wrong values for -B in command line.

gadjex, 2015-07-27:
Select USBasp from Programmer list. select a bit clock and observe incorrect values on -B in in the command line output at the bottom. Also there is no 16 KHz or 32 KHz. There is however 46.88 KHz and 20.96 KHz which is not in the drop down list but show up when selecting 16 KHz and 32 KHz respectively.

zkemble, 2015-07-27:
Hey gadjex, the -B values are correct for the USBasp. The value passed to avrdude is supposed to be the number of microseconds for each clock cycle, avrdude then rounds it down to the nearest valid frequency. So -B 46.88 is ~21KHz which rounded down to the nearest valid frequency is 16KHz. I can't remember why I chose weird values like 46.88 and 20.96 for 16 and 32KHz, but they work.

closed 2015-07-27 by zkemble

gadjex, 2015-07-27:
Yes they do work correctly. The just display incorrect -B values in the grayed out output command line textbox at the bottom. I can have a look at the source and but off the top of my head it seems the array of values linked to the dropdown are sorted backwards. Example I choose 1.5 MHz and in the greyed out textbox it display "... -B 0.5 ...". Likewise if I choose 500 Hz it shows "... -B 1500.0 ...". The values passed to AVRDUDE.EXE are correct, just not the display in the grayed out textbox. I was using the command line and referencing AVRDUDESS as to what commands I needed to enter in a hurry and realized this.

[image]