ZakKemble/AVRDUDESS #5: Using the flashFile parameter in preset.xml file

CurtisFissel, 2016-01-02:
Hello!

I have one of my own presets made... But I cant get the flashFile parameter to work.

Here is what I am trying... I'm sure I am missing something simple.
`<flashFile />C:\firmware.hex</flashFile>` 

Thank you for any help.

Curtis

zkemble, 2016-01-02:
Hi, that first bit with `<flashFile />` is short for `<flashFile></flashFile>` (empty parameter) so with yours you've ended up with `<flashFile></flashFile>C:\firmware.hex</flashFile>`.
Try this `<flashFile>C:\firmware.hex</flashFile>`.

Also, you would normally setup your preset using the AVRDUDESS program rather than manually editing the presets.xml file.

CurtisFissel, 2016-01-02:
Thank you very much. Got it figured out :)

closed 2016-01-02 by CurtisFissel