ZakKemble/AVRDUDESS #7: Needs entry for Arduino as ISP sketch?

mralexgray, 2016-03-01:
What should we select?  `Arduino`? `Atmel AVR ISP`? `Atmel STK500`?  

All are true.. None seem to work.

`Arduino` and `Atmel STK500` do seem to communicate with the programmer, but incorrectly report the ACTUAL arduino's chip, the `atmega328p` when using <kbd>Detect...</kbd>.  Should that be opened as a separate issue, maybe?

zkemble, 2016-03-05:
The Arduino IDE passes `stk500v1` as the programmer to avrdude, which is `Atmel STK500 Version 1.x firmware` in AVRDUDESS. 'stk500' and `avrisp` should also work since they're just aliases of each other. I don't think this is a AVRDUDESS issue really, since it's just a GUI front-end for AVRDUDE. I'd suggest trying to get the Arduino working as an ISP with just avrdude first, since there's plenty of stuff about that on the net already.

mobifu1, 2016-10-15:
i think it is not a problem of sourcecose, a tooltip or comment would be good.
It can spare time to search for the right adjustments.

mike-ady, 2020-04-13:
The "Arduino as ISP" feature (still?) isn't available as of v2.11.  To get the correct settings, I relied on the AVRDude command line in the Arduino IDE.  To use an Arduino running ArduinoISP to program an AVR chip successfully, the following AVRDude command line switches are required:  

-c stk500v1 -b 19200

To obtain these settings, select "Atmel STK500 Version 1.x firmware" under Programmer, and enter "-b 19200" under "Additional command line args".  See attached screen print.

[image]

zkemble, 2020-04-13:
"Arduino as ISP" is one of the default presets since v2.5, though it can be deleted in the preset manager. It sets the programmer to "Arduino". The "-b 19200" doesn't need to go into additional args, just put 19200 into the baud rate under the programmer selection (choosing the "Arduino as ISP" preset already does that too).

mike-ady, 2020-04-13:
Thanks.  I tried clicking on the preset buttons and confusion ensured.  A short description of each control in the ReadMe file would be appreciated.  The "Arduino as ISP" preset also sets the Port to "COM1" unhelpfully.