ZakKemble/AVRDUDESS #14: Simplified Chinese Translation

radiolee, 2018-11-16:
I'm sorry I am new in github, don't know how to pull the file :-( 
Attach the translate file from english to Simplified Chinese (zh-CN), hope you like it.
[chinese-zh_CN.zip](https://github.com/zkemble/AVRDUDESS/files/2587747/chinese-zh_CN.zip)

zkemble, 2018-11-19:
Thanks for this, @radiolee ! Though, since the release of v2.6 a few more translations have been added, highlighted green here - https://github.com/zkemble/AVRDUDESS/compare/v2.6...master#diff-f5b312806011644cec80095d58ae06f2

Could you translate those?

radiolee, 2018-11-21:
done. 
[chinese-zh_CN.zip](https://github.com/zkemble/AVRDUDESS/files/2602410/chinese-zh_CN.zip)

zkemble, 2018-11-21:
Added, will be included in the next release 👍

closed 2018-11-21 by zkemble