ZakKemble/AVRDUDESS #18: Update .NET version

E3V3A, 2019-01-22:
There is no sane reason to keep using the horribly outdated .NET <3.5. The reason is that any .NET above, should be compatible with whats below. At least this is true in more recent versions, since there are some methods to allow to use .NET < using xml config files. TBH. I have never done this, but have seen it done here and there. 

BUT. I do know, that no sane person want to pollute their system by having to install anything 3.5 or less. So please consider updating the app. :+1:

zkemble, 2019-01-23:
MinGW is for C/C++, right? It would take waaaaayyy too much time and effort to port everything to that, while still maintaining cross-platform compatibility.

E3V3A, 2019-01-29:
Yeah, removing .NET may not be possible at this point. But while we investigate, it should be quite straight forward for you to update to use .NET 4.5.  Please consult: 

* https://docs.microsoft.com/en-us/dotnet/framework/deployment/initialization-errors-managing-the-user-experience
* https://docs.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-configure-an-app-to-support-net-framework-4-or-4-5

---

**HEY THIS FIXED IT!!**    Yaayh!! :1st_place_medal:  :tada:  :heart: 

**Add a file called**: `avrdudess.exe.config` with the content: 
```xml
<configuration>
    <startup>
        <supportedRuntime version="v4.0"/>
        <supportedRuntime version="v2.0.50727"/>
    </startup>
</configuration>
```

E3V3A, 2019-01-29:
Funny enough, there's a file called [App.config](https://github.com/zkemble/AVRDUDESS/blob/master/src/avrdudess/App.config) which should have similar content...but only has 2.0.xxxx.

zkemble, 2019-01-29:
That one line seems to have done the trick! I didn't know it could load multiple frameworks depending on what's installed :O Though, there's also a TargetFrameworkVersion / TargetFrameworkVersions option in avrdudess.csproj. I'll have to experiment a bit with a system that only has .net 2.0 installed and see what happens with TargetFrameworkVersion at 4.0 and supportedRuntime as 2.0 and other things. Thanks!

zkemble, 2019-02-08:
Ok, so the supportedRuntime stuff works great on a .net 2.0 system! However, targeting a newer framework doesn't work too well (kinda expected there, tested on Windows XP):
4.6.1:  "Not a valid windows program"
4.0: "Encountered a problem and needs to close"
3.5: Works, but if any new 3.5 stuff is used (like LINQ to XML) then the JIT debugger will pop up with a bunch of errors about missing references and things.

I also can't seem to get multiple framework targets working either, seems to default to 4.0.

So we'll keep the target framework as 2.0 so no newer things are added by mistake, and add the supportedRuntime line so older .net frameworks don't need to be installed on to newer systems :D

E3V3A, 2019-02-09:
great! :+1:

closed 2019-02-09 by zkemble