ZakKemble/AVRDUDESS #21: Feature Request: Read Board Voltage

jacky4566, 2019-02-07:
It would be nice if there was a button to read board voltage like in Atmel Studio.

DirtyEngineer, 2019-02-07:
Not really possible as neither Avrdude or most programming hardware has the ability to do this.

jacky4566, 2019-02-07:
It should be possible though. Atmel Studio can do it with my ICE so the hardware is there. Just a software thing.

I thought you could do it with `-t parms` but it only returns the signature.
So really this needs to be implemented in AVRDUDE first. Just figured `-t parms` would work on ICE...

zkemble, 2019-02-08:
Yea, reading the board voltage would be an avrdude thing, and if that doesn't support it then avrdudess won't either since we only parse the output of avrdude. I think the -t option is for entering terminal mode, where you can manually edit individual bytes in flash and things.

closed 2019-02-08 by zkemble