ZakKemble/AVRDUDESS #22: Feature Request: Program Sequential Serial Numbers to EEPROM

DirtyEngineer, 2019-03-12:
While programming batches of boards I've often needed to give each board a sequential serial number by burning the S# value to a location in the chips EEPROM.

While searching to figure out how to do this most effectively I happened across [this program](https://www.megunolink.com/documentation/program-device/give-devices-unique-ids/). It has a feature that allows you to write unique ids to a boards EEPROM.  Unfortunately it has a very limited number of programmers that it supports and I found the user interface really clunky.

I then discovered that AVRDUDE actually has a "memory operation" command that allows writing a value to memory including EEPROM.

> -U memtype:op:filename[:format]     ([reference manual](http://www.nongnu.org/avrdude/user-manual/avrdude_4.html))

Using AVRDUDESS's "Additional command line args" input I can write a serial number(s) to the boards eeprom while programming the chip. The only limitation is that you cannot specify the location it writes the value to. It starts at the first location and if additional values are included it writes them to subsequence locations.

Example;
> -U eeprom:w:1,2,3:m

Yields an EEPROM download with a first line value of
> :20000000010203FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7

This works great except that the serial number has to be incremented manually each time a board is programmed.

Is it possible to add an input box(s) to AVRDUDESS that when selected increment the number "x" in the command -U eeprom:w:x,x,x:m each time the "Program!" button is pressed?

zkemble, 2019-03-16:
Hmm, automation is kind of out of the scope for this project. I wouldn't think it would be too difficult to create a small script or program to generate a simple .hex file with the memory location and incrementing value everytime it is ran, then avrdude can use the generated .hex file.

DirtyEngineer, 2019-03-17:
Thanks for the reply. I understand your point about the scope of this project. Am I remembering correctly that you mentioned at one point that you hoped to added the ability to trigger an external program from AVRDUDESS? This might be a better way to handle my serial number issue. Is this still on the road map?  

Thank you for providing the source for this project. I've been playing around and learning lots.  I'd like to try to add a button to read the calibration byte(s) from the MCU ( -U calibration:r:filelocation:h ).

zkemble, 2019-03-17:
Yup, I plan on adding the ability to run a program before and after running avrdude. It's high on the list, but I'm not sure when it will be added.

closed 2019-03-21 by DirtyEngineer