ZakKemble/AVRDUDESS #26: Improve MCU auto-detect

zkemble, 2019-09-07:
MCU auto-detect doesn't always work, depending on the programmer and MCU programming interface.
Auto-detect works by running avrdude with the chosen programmer and MCU set to ATmega8, since avrdude needs to have an MCU set before it will do anything. It then reports the device signature which we use to work out what MCU we're really talking to.
But this doesn't always work, for example, if an ATtiny10 is connected to an USBasp then auto-detect will fail because it is expecting an MCU with the usual SPI programming interface, but the ATtiny10 uses TPI.
This can probably be fixed by going through a list of MCUs that have different interfaces until one works and reports its signature. ATmega8 (SPI) -> ATtiny10 (TPI) -> ATxmega16A4U (PDI) -> DebugWire? -> JTAG?

Ser9ei, 2021-04-06:
may be a drop down "listbox" with list of interfaces can help us.
(with spi as default value)

EclipseBoom, 2021-09-16:
I downloaded the latest AVRDudess.  The CPU list doesn't include these late model Tiny parts.  I had to manually update to the latest AVRDude and .conf files.  Then it listed the CPU but even when manually picking the CPU from the list it doesn't put the proper -p in there.  At least that was my experience.

zkemble, 2021-09-16:
@EclipseBoom Yes, that is expected. Auto-detect tries to automatically detect the MCU for you, but it only works with parts that use the SPI programming interface. Selecting an MCU first will not change how auto-detect works.

EclipseBoom, 2021-09-16:
OK that seems odd.  Why can't you force the CPU choice?  If I accidentally put in the wrong CPU say from an old version of hardware I don't want the software to autodetect that CPU and override my choice and just use it.  

If I choose a CPU I want it to error out if I put connect a different one to the programmer not just change the CPU setting and program away.  This is what MicrochipStudio and AVRStudio do.  This is an actual case with ATMega168PB vs ATMega168PA chips.  The PB chips don't work and the PA do.  So if I mistakenly mount a PB I don't want to program it and have it verify etc.  I also have some ATtiny1616 mounted and some ATtiny1626.  Same PCB different CPU that isn't software compatible but you can load the HEX file for the 1626 into the 1616.

zkemble, 2021-09-16:
The whole idea of auto-detect is to simply detect what MCU you have your programmer connected to and automatically select it from the drop-down list. If you don't want it to override your chosen MCU then don't click on the auto-detect button. If you have a different microcontroller connected from the one you selected it will return an error when you attempt to program or read it.

EclipseBoom, 2021-09-16:
MicrochipStudio has a button for "Read" Device Sig that will do a check to see if the attached CPU matches the chosen unit.

zkemble, 2021-09-16:
Ok, so you're asking for a new feature that only verifies that the connected microcontroller is the correct part. I'll add it to the todo list, but I've no idea when it'll be added.

harryberlin, 2021-09-16:
i mean avrdude should do this already.
check the args for avrdude.

zkemble, 2021-09-16:
Yes, avrdude can read just the signature if only the programmer and MCU parameters are passed to it (that's how the auto-detect feature works), but AVRDUDESS still needs a new button or something for the feature requested by @EclipseBoom where it just makes sure that the microcontroller matches the selected part.

zydl123, 2022-01-17:
I used the ft245bl programmer, but the software couldn't find it
[FT245BL.txt](https://github.com/zkemble/AVRDUDESS/files/7880145/FT245BL.txt)

fsmoke, 2022-10-28:
Also i want to add - better use t13 but not m8 MCU for detect - because m8 not support HVSP programming mode see log
```
avrdude.exe: stk500hv_initialize(): high-voltage serial programming control stack not defined for part "ATmega8"
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.
```
So you must test MCU 3 times - for TPI interface, for HVSP interface(like tiny13) and HVPP Interface(like mega8)

rompipelotas, 2024-05-03:
is it possible with avrdudess to read lgt8f328??