ZakKemble/AVRDUDESS #27: Update language files

zkemble, 2019-09-10:
Language files need to be updated with translations for new strings. Currently French, Russian, Polish and Chinese.

Protoncek, 2020-06-20:
Hello!
I use this program quite a while now, so i took the time and translated it to Slovenian language, if it's usefull...

thanks for your work!
[slovenian.zip](https://github.com/zkemble/AVRDUDESS/files/4807701/slovenian.zip)

zkemble, 2020-06-22:
Awesome thanks! Though, there seems to be a few missed translations -
TOOLTIP_LOWFUSE
TOOLTIP_HIGHFUSE
TOOLTIP_EXFUSE
BITCLOCK
FORMAT
VERBOSITY
AVRDUDEKILLED
UPDATE_BADXML
FILEFMT_BIN

Protoncek, 2020-06-22:
Oopss... 
i missed some, yep... I corrected those. Word "Format" is exactly the same in Slovenian language, so i left it untouched. rest of them i translated best way i can...

BTW.... is it normal that if i have XMEGA128A1U then clicking on "detect" doesn't work? Rest is working fine, only detection doesn't... so no big deal, just asking...

Best regards!
[slovenian.zip](https://github.com/zkemble/AVRDUDESS/files/4813735/slovenian.zip)

zkemble, 2020-06-22:
Added! Translations will be in the next release.

Yeah, detect doesn't always work depending on the MCU and programmer combination. Not a bug, just a limitation at the moment. See #26

trudK45, 2020-12-17:
Looks like the german language file is missing. I am not able to build.

Severity	Code	Description	Project	File	Line	Suppression State
Error		Could not copy the file "C:\Users\hasse\Sources\AVRDUDESS\src\avrdudess\Languages\german.xml" because it was not found.	avrdudess

zkemble, 2020-12-17:
> 
> 
> Looks like the german language file is missing. I am not able to build.
> 
> Severity Code Description Project File Line Suppression State
> Error Could not copy the file "C:\Users\hasse\Sources\AVRDUDESS\src\avrdudess\Languages\german.xml" because it was not found. avrdudess

Oops, now fixed in b86d961f1a53c74e0d1c1f102cba9951447c1103

The translation file was removed since it was still in English and only for testing, then I forgot to update the .csproj file. Will add back once someone translates to German.

closed 2024-01-05 by ZakKemble