ZakKemble/AVRDUDESS #28: JIT error with 2.10 version

Bastiaang, 2019-10-11:
Just updated AVRDUDESS from 2.8 to 2.10; When I am starting up AVRDUDESS 2.10 an error appears with the following text:

************** Tekst van uitzondering **************
System.NullReferenceException: De objectverwijzing is niet op een exemplaar van een object ingesteld.
   bij avrdudess.CmdLine.addWriteFuses()
   bij avrdudess.CmdLine.generate()
   bij avrdudess.Form1.event_controlChanged(Object sender, EventArgs e)
   bij System.Windows.Forms.CheckBox.OnCheckedChanged(EventArgs e)
   bij System.Windows.Forms.CheckBox.set_CheckState(CheckState value)
   bij avrdudess.Form1.set_setFuses(Boolean value)
   bij avrdudess.Form1.loadPresetData(PresetData item)
   bij avrdudess.Form1.Form1_Load(Object sender, EventArgs e)
   bij System.Windows.Forms.Form.OnLoad(EventArgs e)
   bij System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   bij System.Windows.Forms.Control.CreateControl()
   bij System.Windows.Forms.Control.WmShowWindow(Message& m)
   bij System.Windows.Forms.Control.WndProc(Message& m)
   bij System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   bij System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

(I am sorry that I am using the dutch version)

I can choose between continue or shutdown. When I continue the interface of AVRDUDESS is a mess.
After reinstalling AVRDUDESS 2.10 I am still getting the same error. When I downgrade to 2.8 there is no error anymore.

zkemble, 2019-10-11:
Thanks for reporting this! Seems this is caused by having no MCU selected and ticking the "Set fuses" box. I'll get it fixed this weekend.

closed 2019-10-12 by zkemble

Bastiaang, 2019-10-14:
The update 2.11 has fixed my issue. Thanks for the fast reply and fix!