ZakKemble/AVRDUDESS #29: bit clock (-B)

MattBelle95, 2019-11-21:
i was trying to upload a hex file to test my atmega32, but i forgot to set the fuse bits for clock source selection, the code uploaded just fine but the delay function was much slower (i was using 12MHz crystal and the default settings was the internal 1MHz), i edited the fuse bits and uploaded the code, nothing bad so far, but when i tried to upload it again it gave me an error rc=-1 and that it can't set clock bit to specified value which was 1.5MHz.
is there a value i should use for clock bits (-B) corresponding to the selected frequency ?

zkemble, 2019-12-18:
Hey Matt, you've probably got this fixed now, but did you also set the divide clock by 8 fusebit?

MattBelle95, 2019-12-28:
@zkemble Hey Zak, I didn't get this problem fixed yet, I am using atmega32 it doesn't have divide by 8 fuse bit
[image]

zkemble, 2019-12-28:
Have you updated the USBasp firmware (I'm guessing thats the programmer you're using)? They usually come with old firmware where the bit block can only be changed by a jumper on the programmer.

Also send the console output of the error and the new fues values that you tried.

MattBelle95, 2019-12-30:
I can't get the console output of the error because i bought a new controller.
i used this calculator to get fuse bits values [http://www.engbedded.com/fusecalc](url)
Lfuse:0xFF  Hfuse:0x89

MattBelle95, 2019-12-30:
I found my old broken controller
here are the screen shots of the error
[image]

and this when i checked -F
[image]

zkemble, 2019-12-30:
It says in the error message "please check for usbasp firmware update", so looks like the programmer firmware needs updating.

MattBelle95, 2019-12-30:
so the problem is in the usbasp programmer?
where can i get the trusted latest firmware update?

DirtyEngineer, 2019-12-31:
@MattBelle95 https://www.fischl.de/usbasp/

MattBelle95, 2019-12-31:
@zkemble i will download the driver and let you kniw what happened, thanks a lot for your effort

MattBelle95, 2020-01-11:
@zkemble i downloaded the latest version (i already had it installed, but i removed it and installed it again).
it still gives me the same error.
another thing i noticed is when i upload a hex file to my new atmega32 (it uploads successfully and the code behaves as expected) it gives me a warning that it can't set SCK frequency

[image]

zkemble, 2020-01-11:
I think you only did something with the driver (libUSB), not the firmware. You need a second programmer, like another USBasp or an Arduino with the ArduinoISP sketch loaded onto it to re-program your USBasp. Do a google search for "usbasp firmware update", there are loads of tutorials on how to do it.

There is also usually a jumper on the USBasp somewhere to select fast (1.5MHz) or slow (375KHz) clock (only for the old firmware, new firmware allows setting the clock by software, which is what the error is all about), try shorting or opening it to see if it can then talk to the bad ATmega32.

closed 2020-05-19 by zkemble