ZakKemble/AVRDUDESS #31: Inno Setup 6 update

mike-ady, 2020-04-13:
I built AVRDudess from the latest Github source.  The .exe built without issue.  

I did not have Inno Setup, so I installed the latest version.  I had to modify installer/installer.bat to reflect the version change from 5 to 6.  I also had to modify installer/avrduddess.iss to comment out several languages which are apparently not supported (yet?) in Inno Setup 6:

;Name: "greek"; MessagesFile: "compiler:Languages\Greek.isl"
;Name: "hungarian"; MessagesFile: "compiler:Languages\Hungarian.isl"
;Name: "scottishgaelic"; MessagesFile: "compiler:Languages\ScottishGaelic.isl"
;Name: "serbiancyrillic"; MessagesFile: "compiler:Languages\SerbianCyrillic.isl"
;Name: "serbianlatin"; MessagesFile: "compiler:Languages\SerbianLatin.isl"

zkemble, 2020-04-13:
Thanks! I'll get things updated to Inno v6 for the next release.

zkemble, 2020-05-19:
Just an FYI; Inno Setup 6 installers no longer support Windows 2000, XP, and Server 2003. Vista is the minimum.

closed 2020-05-19 by zkemble