ZakKemble/AVRDUDESS #35: Manjaro AVRDUDESS

Samuel-ZDM, 2020-07-25:
Hello!

I am trying to use AVRDUDESS in Manjaro. I installed mono, but when I use the mono command, mono AVRDUDESS-2.13-setup.exe, the following error occurred:

The set 'AVRDUDESS-2.13-setup.exe' cannot be opened: the file does not contain a valid CIL image.

How can I correct that?

Thank you!

zkemble, 2020-07-25:
AVRDUDESS-2.13-setup.exe is an installer for Windows. Use the portable version instead.

Samuel-ZDM, 2020-07-25:
Thank you. It Worked!
Thank you. It worked!

Now, when I opened it, it showed

>>>: avrdude
ERROR: avr size is missing!

What would that be? Why don't I have the card connected?

zkemble, 2020-07-25:
Awesome :) You'll need to install AVR-GCC to get avr-size. It's only used to show a little blue memory usage bar.

Samuel-ZDM, 2020-07-25:
I installed and for now without error. I will try to pass the code. 
Anything return.

Thank you very much

closed 2020-07-28 by Samuel-ZDM