ZakKemble/AVRDUDESS #36: Arduino Uno ISP

Samuel-ZDM, 2020-10-29:
Hello!

I am using Arduino Uno as an ISP to record to an Atmega328.
I recorded the ISP code on Arduino Uno.
My programmer is the Atmel AVR ISP and is giving a signature error when I try to detect the board.

Am I using the correct programmer method?

Thank you very much in advance

[image]

zkemble, 2020-10-29:
Choose "Arduino" as the programmer. There's also a preset (right side) called Arduino as ISP.

Samuel-ZDM, 2020-10-29:
I did that, but the same mistake happened.
Then, I removed the baud rate and instead of recording on Atmega328, he recorded on Arduino Uno.

Samuel-ZDM, 2020-10-29:
I was trying here and put the programmer as Atmel AVR ISP and Presets with Arduino as ISP. It looks like it worked.

closed 2020-10-30 by Samuel-ZDM