ZakKemble/AVRDUDESS #38: Does not detect USBasp

Samuel-ZDM, 2020-11-12:
Hello!

I have a USBasp recorder, but AVRDUDESS does not recognize it. 
I tried avrdude on the command line and it seems to work.
I added the file to the rules and it still doesn't work.

Thanks in advance

zkemble, 2020-11-12:
I'm assuing you're using Linux since you mentioned rules files. What rules did you add? Have you tried running AVRDUDESS as root?

Samuel-ZDM, 2020-11-12:
I added in a file named 99-USBasp.rules:

> SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", SYSFS{idVendor}=="0x16c0", SYSFS{idProduct}=="0x05dc", MODE="0666"
> SYSFS{idVendor}=="16c0", SYSFS{idProduct}=="05dc", MODE="0660", GROUP="plugdev"
> ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="05dc", GROUP="usb", MODE="0666"

Yes. I tried with sudo mono avrdudess.exe

zkemble, 2020-11-12:
Could you paste the output of AVRDUDESS when it fails to find the USBasp?

Samuel-ZDM, 2020-11-12:
I can't send the code, because the AVRDUDESS doesn't show the Port.
[image]
Very strange that.

zkemble, 2020-11-12:
The USBasp is not a serial port. From the programmer drop down box go to the very bottom and choose USBasp.

Samuel-ZDM, 2020-11-12:
Thank you!

I believe it is now working.
Only at the end, at the moment of reading, is he giving a mismatch error, but I believe he is recording.

Error:

> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x0000
>           0x00! = 0x0c
> avrdude: verification error; content mismatch

Samuel-ZDM, 2020-11-17:
Is not working. I can do the readings of the Fuses, the detection of the device, but when I record it it gives this error at the end and the device does not work. What can this be?

DirtyEngineer, 2020-11-17:
Try selecting the "Erase flash and EEPROM (-e)" option. The lockbits might be set on your chip. Also remove the "Force(-F)" option.

Samuel-ZDM, 2020-11-17:
Thanks!

Did not work.
I have already tried to add a resistor between RESET and VCC and it didn't work either.

DirtyEngineer, 2020-11-17:
If Avrdude is reading the chip ID and fuses it is communicating with the chip. Please turn on Verbosity 4 and post the full output.

Samuel-ZDM, 2020-11-17:
I removed some lines of output from the moment of writing on the device, because the output exceeded the maximum number of characters allowed in the comment.

avrdude: Version 6.3, compiled on Jul  7 2020 at 19:38:43
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Overriding Baud Rate          : 9600
         Setting bit clk period        : 0.5
avrdude: usbasp_open("usb")
avrdude: seen device from vendor ->www.fischl.de<-
avrdude: seen product ->USBasp<-
         AVR Part                      : ATmega328
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

[... 871 lines cut ...]
avrdude: avr_read(): skipping page 239: no interesting data
avrdude: avr_read(): skipping page 240: no interesting data
avrdude: avr_read(): skipping page 241: no interesting data
avrdude: avr_read(): skipping page 242: no interesting data
avrdude: avr_read(): skipping page 243: no interesting data
avrdude: avr_read(): skipping page 244: no interesting data
avrdude: avr_read(): skipping page 245: no interesting data
avrdude: avr_read(): skipping page 246: no interesting data
avrdude: avr_read(): skipping page 247: no interesting data
avrdude: avr_read(): skipping page 248: no interesting data
avrdude: avr_read(): skipping page 249: no interesting data
avrdude: avr_read(): skipping page 250: no interesting data
avrdude: avr_read(): skipping page 251: no interesting data
avrdude: avr_read(): skipping page 252: no interesting data
avrdude: avr_read(): skipping page 253: no interesting data
avrdude: avr_read(): skipping page 254: no interesting data
avrdude: avr_read(): skipping page 255: no interesting data
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0x0c
avrdude: verification error; content mismatch
avrdude: usbasp_close()
avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude done.  Thank you.

DirtyEngineer, 2020-11-17:
It's not writing to the chip, what are your fuse and lock bits set to?

Also you have a clone with older usbasp firmware that doesn't allow a bit clock setting.  Try using the "Any usbasp clone with correct VIP/PID" in the programmer setting.

Do you have another chip you can try?

Samuel-ZDM, 2020-11-17:
My configuration is this:
[image]

I tried with "Any USBasp clone with correct VIP / PID" and changed only the HEX numbers:

> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x0042
>          0xf6 != 0xf7
> avrdude: verification error; content mismatch

I also have another card with Atmega328. I can try this one.

zkemble, 2020-11-17:
If the chip was locked then reading it would only return 0xFF values. However, here it looks like it returns the word address instead of the data at that address.

I don't think choosing "Any usbasp clone..." will help here, the only difference between that and the normal USBasp option is that it doesn't check the vendor and product strings when searching for the USB device. But yes, you should update the firmware on your USBasp, which will require a second programmer (another USBasp or an Arduino as ISP).

What is the verbose output when choosing "Any USBasp clone..."? Since it looks like the first mismatch is different.

DirtyEngineer, 2020-11-17:
Also try "Read" on both the Fuses and Lock bits.

Samuel-ZDM, 2020-11-17:
@DirtyEngineer  I can read the fuses. It show the values I put.

Samuel-ZDM, 2020-11-17:
@zkemble I managed to get the detailed output, but it showed the same error. I don't know why the other one has changed. But it is the same error that is appearing.

> vrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x0000
> 0x00 != 0x0c
> avrdude: verification error; content mismatch
> avrdude: usbasp_close()
> avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

ahmad-wahyudin, 2020-11-17:
I have same problem with Linux, Opensuse Leap 15

zkemble, 2020-11-17:
Ok, this is not an AVRDUDESS issue. The next step would be to update the USBasp firmware, googling for "update usbasp" will bring up loads of tutorials on how to do it. Closing issue for now.

closed 2020-11-17 by zkemble

Samuel-ZDM, 2020-11-17:
@zkemble Sure. I'm sorry for this.

Thank you very much for your help.