ZakKemble/AVRDUDESS #43: New action/button -

Ser9ei, 2021-03-31:


Ser9ei, 2021-04-01:
I mean add separate action to execute avrdude.exe with '-e' parameter (Perform a chip erase).

zkemble, 2021-04-02:
Seems like a good idea, I'll get it added.

7918514, 2022-07-12:
> Seems like a good idea, I'll get it added.

Hello. Forgot about the button?

ZakKemble, 2022-07-12:
I've not forgotten, it's still on the todo list. I'm not sure when I'll get it added though.

7918514, 2022-07-30:
How about these buttons? 

[image]

[image]

ZakKemble, 2022-07-30:
I'm not sure what Empty is supposed to do? There's a Delete button at the bottom.

7918514, 2022-07-30:
This is for separation. Separator between groups. Like in the screenshot.

7918514, 2022-07-30:
Here's another idea))) To move presets.

[image]

ZakKemble, 2022-07-30:
The present manager isn't very good at the moment. I need to completely redesign it sometime. u.u