ZakKemble/AVRDUDESS #44: "Fuse && lock bits: Atmea32u4" window - Warning: Unsuported MCU

Ser9ei, 2021-04-01:
Add support (or correct warning notification) of the Atmea32u4 MCU in the "Fuse && lock bits: Atmea32u4" window.
Please see screenshot in the attach.

[image]

Ser9ei, 2021-04-01:
I believe the "bits.xml" file should be update by text below:

	<mcu signature="1e9587" name="atmega32u4">
		<high>BOOTRST,BOOTSZ0,BOOTSZ1,EESAVE,WDTON,SPIEN,JTAGEN,OCDEN</high>
		<low>CKSEL0,CKSEL1,CKSEL2,CKSEL3,SUT0,SUT1,CKOUT,CKDIV8</low>
		<ext>BODLEVEL0,BODLEVEL1,BODLEVEL2,HWBE,,,,</ext>
		<lock>LB1,LB2,BLB01,BLB02,BLB11,BLB12,,</lock>

zkemble, 2021-04-02:
Alright, I'll add those fuse bit settings for the next release. You can submit a pull request instead if you want to.

closed 2022-05-09 by ZakKemble