ZakKemble/AVRDUDESS #45: www.engbedded.com website not available & "Fuse settings" feature/link does't work

Ser9ei, 2021-04-02:
The best solution for resolve issue is add the ability to modify link to the website in text (xml) configuration file (for emxample, in the config.xml file).
Yes, it more heavy case, in this case pattern of the link and arguments (fuses) should be described in the configuration file also.
Or hardcore - change website on code source.

zkemble, 2021-04-02:
The engbedded.com database error seems to be a recent thing. Hopefully they'll get it fixed. An alternative site is https://eleccelerator.com/fusecalc/ but it doesn't seem to have as many MCUs to choose from. Really, I should get the fuse calculator that's already built in to AVRDUDESS updated so we don't have to rely on external stuff.

Ser9ei, 2021-04-02:
Integral calculator is an awesome solution.

ro-bercik, 2022-02-21:
Site: https://www.engbedded.com/fusecalc/ works again from some time, there is also new version from the same author: https://www.engbedded.com/conffuse/ . And as a "special" alternative: https://www.vagrearg.org/content/atpack (script which parse atdf files from Microchip packs).

ZakKemble, 2022-05-09:
Reverted changes 53111c028b84f86aa7536f43d5652aa2fa2116aa

closed 2022-05-09 by ZakKemble