ZakKemble/AVRDUDESS #50: Avrdude is missing!

Samuel-ZDM, 2021-05-25:
Hello!

I am trying to run avrdudess in manjaro, but it is giving this error: Gtk not found (missing LD_LIBRARY_PATH to libgtk-x11-2.0.so.0?), Using built-in colorscheme.

I don't know how to solve this, I'm not able to install this missing package.

Thank you very much in advance!

Samuel-ZDM, 2021-05-25:
I managed to fix it. But the Avrdude message is missing continues. I have already installed the AVR GCC.

zkemble, 2021-05-25:
Avrdude is its own package. `pacman -S --needed avrdude` or whatever the equivalent is for your OS.

Samuel-ZDM, 2021-05-25:
Thank you very much!

I executed the command and the error disappeared!
Thank you very much again, I didn't know what else to do. I will test to see if I can record, but I believe it will work.

See you.

closed 2021-05-25 by Samuel-ZDM