ZakKemble/AVRDUDESS #55: Usbasp not recognized, but Arduino IDE will upload with it.

jtannenba, 2021-09-18:
I have one of these:  https://www.ebay.com/itm/263711477076 and Avrdudes complains with:  avrdude.exe: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'. Is this a problem with Avrdudess or Avrdude?  The Arduino IDE will use this to upload sketches ok.

Thanks,
Joe

zkemble, 2021-09-19:
Hi, use [Zadig](https://zadig.akeo.ie/) to make sure it's using the libusb-win32 driver. I think the Arduino avrdude supports both WinUSB and libusb-win32 drivers which is probably why it's working there.

jtannenba, 2021-09-19:
That did it.  Was listed in Device manager as Usbasp, now as libusb-win32 device. and yes, now Avrdudess recognizes it.  
Thank you,
Joe

closed 2021-09-19 by zkemble

jtannenba, 2024-01-20:
Zack,
I really like avrdudess and use it quite oftern.  Upgraded to 2.15 and when clicking detect when I'm connected to a atmega328, it comes back as the ata6614Q and not the atmega328.  Is this a bug or a feature?
Thanks,joe

ZakKemble, 2024-01-20:
Hmm, technically, it's not a bug or a feature lol. The ATA6614Q is a combo chip that contains an ATmega328. However, a fix will be in v2.16 to ignore ATA661x parts since most people will be using the normal standalone parts rather than the combo chips. See here https://github.com/ZakKemble/AVRDUDESS/issues/81

jtannenba, 2024-01-20:
Thank you, that explains it.
Joe