ZakKemble/AVRDUDESS #56: Push flash in atemga16

nghiank97, 2021-09-25:
I have a project in **atmega16**. I using **USP ISP** to push the hex file. But, I don't know how to setup. Please help me

[image]
[image]

DirtyEngineer, 2021-09-25:
You need to select the correct programmer.  In the top left drop-down select either "USBasp, http://www.fischl.de/usbasp/" or "Any usbasp clone with correct VID/PID". Next click the "Detect" button on the top right side. If you have made the correct connections the lower text box will show the signature/name of your chip. Then click the program button.

nghiank97, 2021-09-25:
**Thank for your response**

I have try it, but i don't
When I select **"USBasp, http://www.fischl.de/usbasp/"** and press "Detect". The commend show **"could not find USB ...."**

[image]

This is the same when I select **"Any usbasp clone with correct VID/PID"**

But when i select **"Atmel AVR ISP"**

[image]

DirtyEngineer, 2021-09-25:
Does the programmer show up in your Device Manager? If not you probably need to download the drivers for it. Goggle should be able to help with how to get it to work.

nghiank97, 2021-10-02:
ok, many thanks

closed 2022-05-09 by ZakKemble