ZakKemble/AVRDUDESS #60: AVRDUDE 6.4 \o.o/

zkemble, 2022-02-03:
6.4 was released a few months ago,but it's missing libhidapi-0.dll.
The .dll can be downloaded here - https://github.com/libusb/hidapi/releases/tag/hidapi-0.11.2 then place the x86 version into the AVRDUDESS directory and rename it to libhidapi-0.dll

zkemble, 2022-02-16:
Official avrdude repo: https://github.com/avrdudes/avrdude
Forked repo with v7.0-rc1: https://github.com/mariusgreuel/avrdude/releases/tag/v7.0-rc1

DirtyEngineer, 2022-02-16:
Finally actual Atmel-ICE support! This is wonderful.

ro-bercik, 2022-02-21:
Version 7.0RC1 tested with FT232R (programmer: arduino-ft232r), it works without any problems and without any external dlls.

steigerbalett, 2022-04-21:
AVRDUDE Version: 7.0RC1
Tested with: USBasp
Works without any problems

ZakKemble, 2022-05-09:
Official v7.0 released a few hours ago https://github.com/avrdudes/avrdude/releases/tag/v7.0

pcfreak1201, 2022-12-27:
AVRDUDESS-2.14-portable and avrdude-7.0 (compiled) works on Linux Mint 20.3 Una (base: Ubuntu 20.04 focal), but needed to change to avrdude-7.0.conf for full support - great!

closed 2024-01-05 by ZakKemble