ZakKemble/AVRDUDESS #63: Windows defender warns about virus in the AVRDUDESS-2.14-setup.exe file

Ser9ei, 2022-05-12:
actions:
1. Try to download the AVRDUDESS-2.14-setup.exe file
2. After the downloaded is complete, the Defender warns about virus in the installation file
Please find details in the screenshot below.
[image]

Portable version doesn't have issue with Defender

ZakKemble, 2022-05-12:
Antivirus programs have always hated AVRDUDESS or the installer for some reason :(
Odd that my Defender doesn't see it as a problem. I will see if I can submit a false positive report to Microsoft.

ZakKemble, 2022-05-13:
Microsoft has removed the detection which should be included in the next definition update.

Ser9ei, 2022-05-13:
Thanks, Zak, for the new release

closed 2022-05-13 by Ser9ei