ZakKemble/AVRDUDESS #67: Lock Bits programming problem

7918514, 2022-07-30:
Sorry for my bad english. Google translator.
Thank you very much for your development. I really like your program.
When programming atmega168p/328p, attini45/85 an error occurs when recording Lock Bits. This happens if factory fuses are installed in the microcontroller and I do the firmware of the microcontroller and fuses automatically. This doesn't always happen. Two times out of ten.
[image]

`
~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~
>>>: avrdude -c usbasp -p m168p -P usb -e -B 0.5 -U flash:w:"\\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex":a -U lfuse:w:0xFF:m -U hfuse:w:0xDF:m -U efuse:w:0xF9:m -U lock:w:0xC0:m 

avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e940b (probably m168p)
avrdude.exe: erasing chip
avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: reading input file "\\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex"
avrdude.exe: input file \\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex auto detected as Intel Hex
avrdude.exe: writing flash (4804 bytes):

Writing | ################################################## | 100% 2.78s

avrdude.exe: 4804 bytes of flash written
avrdude.exe: verifying flash memory against \\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex:
avrdude.exe: input file \\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex auto detected as Intel Hex

Reading | ################################################## | 100% 1.40s

avrdude.exe: 4804 bytes of flash verified
avrdude.exe: reading input file "0xFF"
avrdude.exe: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lfuse written
avrdude.exe: verifying lfuse memory against 0xFF:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lfuse verified
avrdude.exe: reading input file "0xDF"
avrdude.exe: writing hfuse (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.03s

avrdude.exe: 1 bytes of hfuse written
avrdude.exe: verifying hfuse memory against 0xDF:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x0000
             0xff != 0xdf
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
`

DirtyEngineer, 2022-07-30:
I've noticed you have added an addition command line argument of -B0.5 (bit clock at 1500000 hrz) which is too fast to program chips with the stock clock settings. This will override your default setting and cause issues.

7918514, 2022-07-30:
> 

It does not help. I remove this setting and set the frequency to 187.5kHz. Error again.

ZakKemble, 2022-07-30:
375KHz is the default frequncy that AVRDUDE uses (or maybe it's the default of USBasp, I'm not sure) if no -B option is supplied.

As for your problem, try enabling verbosity in the Options panel. That might help tell you what's going on.

7918514, 2022-07-30:
I don't know what to watch. Tests have shown that there is no error with verbosity 4. I am attaching the files.

[verbosity_1_error.txt](https://github.com/ZakKemble/AVRDUDESS/files/9226947/verbosity_1_error.txt)
[verbosity_2_error.txt](https://github.com/ZakKemble/AVRDUDESS/files/9226948/verbosity_2_error.txt)
[verbosity_3_error.txt](https://github.com/ZakKemble/AVRDUDESS/files/9226949/verbosity_3_error.txt)
[verbosity_4_NOerror.txt](https://github.com/ZakKemble/AVRDUDESS/files/9226950/verbosity_4_NOerror.txt)

Question about settings manager. All settings are saved. Problem with "bitclock" setting. It is saved in the presets.xml file. But when selecting a saved preset, "bitclock" is not set.

ZakKemble, 2022-07-30:
Hmm, it could be a timing thing. Verbosity 4 might be slowing avrdude down enough that it just about works. Does it work without verbosity and a very slow clock?

Can confirm the bit clock restore problem, I've made an issue here #68

7918514, 2022-07-30:
There are no errors at 93750 Hz.

ZakKemble, 2022-07-30:
Ok, it's a timing thing with AVRDUDE. I think it would be best to make an issue over at https://github.com/avrdudes/avrdude

7918514, 2022-07-30:
Can you ask a question there? And my English is bad. And I don't know many terms/names.

ZakKemble, 2022-07-30:
Your English is fine :) Put a link back to this issue so they can see what's been happening. Title it something like "Fuses sometimes fail to write"

closed 2024-01-14 by ZakKemble