ZakKemble/AVRDUDESS #75: Strange Error when I try to open AVRDUDESS

ParaysteGS, 2023-04-03:
[image]

ZakKemble, 2024-01-07:
Hi, does this still happen with v2.15? https://github.com/ZakKemble/AVRDUDESS/releases/tag/v2.15

wwwGPS, 2024-02-09:
Hi ZAK. I had no problems running AVRDUDESS any version other than 2.16. An error appears when launching AVRDUDESS 2.16. Runs on Win7. Сan you help?

ZakKemble, 2024-02-11:
Hey @wwwGPS, I noticed this error just after releasing v2.16 and fixed in this commit https://github.com/ZakKemble/AVRDUDESS/commit/9cf18b5c98471aae96a7c26ef50fd5fe438e537b

I will be releasing v2.17 in the next few days with the fix.

It's caused by having an old version of .NET Framework. v2.0 or v3.5, I think. v4.0 or newer works fine.

wwwGPS, 2024-02-15:
Thank you! 2.17 works. But there are error "ERROR: Update check failed: The requested security protocol is not supported." in portable version. This is serious problem?

ZakKemble, 2024-02-15:
It's not a problem, it just can't tell you about newer versions when I release them. You can fix the error by updating .NET Framework - https://dotnet.microsoft.com/en-us/download/dotnet-framework/net481

wwwGPS, 2024-02-15:
OK!

closed 2024-09-07 by ZakKemble