ZakKemble/AVRDUDESS #78: AVRDUDE 7.2 - new format of avrdude.conf

ro-bercik, 2023-08-05:
There is a new version of avrdude.exe from official repo: https://github.com/avrdudes/avrdude
But... there was some changing in avrdude.conf config file.
The "id" tag in "programmer" section now contains multiply strings, so code for AvrDudess need some updates.

Here is an example:
7.1) avrdude.conf:
programmer
    id                     = "arduino-ft232r";
    desc                   = "Arduino: FT232R connected to ISP";
    type                   = "ftdi_syncbb";

7.2) avrdude.conf:
programmer
    id                     = "arduino-ft232r", "diecimila";
    desc                   = "Arduino: FT232R connected to ISP";
    type                   = "ftdi_syncbb";

BlackBrix, 2023-11-30:
I second that
+1

ZakKemble, 2024-01-06:
Fixed in this commit https://github.com/ZakKemble/AVRDUDESS/commit/905a19d74852085cc2e04a9077b8ec265b69d51c and now available in v2.15 - https://github.com/ZakKemble/AVRDUDESS/releases/tag/v2.15

closed 2024-01-06 by ZakKemble