ZakKemble/AVRDUDESS #80: Unable to read fuses on atmega328pb with v2.15, v2.14 works perfectly fine

igor-ziuba, 2024-01-06:
As described in title, on new update, while reading fuses from atmega328pb with USBasp, I'm receiving only 

```
Reading fuses...
>>>: avrdude.exe -c usbasp -p m328pb -U hfuse:r:-:h -U lfuse:r:-:h -U efuse:r:-:h 
ERROR: Unable to read fuses/lock bits

avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)

avrdude: processing -U hfuse:r:-:h
avrdude: reading hfuse memory ...
avrdude: writing output file <stdout>

avrdude: processing -U lfuse:r:-:h
avrdude: reading lfuse memory ...
avrdude: writing output file <stdout>

avrdude: processing -U efuse:r:-:h
avrdude: reading efuse memory ...
avrdude: writing output file <stdout>

avrdude done.  Thank you.
```
At the same time, MCU auto-detect works good, and with previous version 2.14 fuse reading works perfectly fine.

```
Reading fuses...
>>>: avrdude -c usbasp -p m328pb -U hfuse:r:-:h -U lfuse:r:-:h -U efuse:r:-:h 
SUCCESS: Read high fuse
SUCCESS: Read low fuse
SUCCESS: Read extended fuse
```

BTW, best GUI for avrdude, love it so much :)

ZakKemble, 2024-01-06:
Hi, it'll be a few days before I can look into this myself. Could you see if reverting avrdude back to v7.0 has any effect - https://github.com/avrdudes/avrdude/releases/download/v7.0/avrdude-v7.0-windows-x86.zip you'll need to replace avrdude.exe and avrdude.conf in the install directory of AVRDUDESS, or go into the options and set the locations there.

ZakKemble, 2024-01-06:
I think the output of avrdude might have changed:

    avrdude error: cannot set sck period; please check for usbasp firmware update

It now says `error` instead of `warning`, so AVRDUDESS thinks there was a problem when reading fuses.

https://github.com/ZakKemble/AVRDUDESS/blob/cb953e3f9644845647a9692b5af9b9836deef3cf/src/avrdudess/Avrdude.cs#L474

Updating your USBasp firmware will sort it out. I'll also add some extra logic so it doesn't fail becuase of the firmware error.

igor-ziuba, 2024-01-06:
> Hi, it'll be a few days before I can look into this myself. Could you see if reverting avrdude back to v7.0 has any effect - [avrdudes/avrdude@`v7.0` avrdude-v7.0-windows-x86.zip (download)](https://github.com/avrdudes/avrdude/releases/download/v7.0/avrdude-v7.0-windows-x86.zip) you'll need to replace avrdude.exe and avrdude.conf in the install directory of AVRDUDESS, or go into the options and set the locations there.

Yeah, works fine, as previously, thanks! Sorry for the inconvenience, I'm understand that you're not responsible for ardude.

closed 2024-01-06 by igor-ziuba

TzOk83, 2024-01-06:
> Updating your USBasp firmware will sort it out.

Which version would you recommend? This seems to be a recent one: https://github.com/dioannidis/usbasp/releases/tag/v1.11

ZakKemble, 2024-01-06:
I've only used the original firmware which you can get from here https://www.fischl.de/usbasp/

commit 64b90cda (2024-01-07): fix #80 reading fuses fails with USBasp old fw