ZakKemble/AVRDUDESS #82: Unable to detect Atmel-ICE in ISP Mode. Right drivers installed

dzalf, 2024-01-13:
For the life of me, I cannot get AVRDUDESS to detect and work with ATMEL-ICE programmer.

[image]

I correctly installed the libusb32 drivers using the Zadig tool.

Interestingly, I am able to program the bootloader and code using platformio.

What is happening?

[image]

ZakKemble, 2024-01-13:
Heya, try deleting the contents in the Port (-P) box, so it's just blank.

dzalf, 2024-01-14:
Hi @ZakKemble 

Thanks for your prompt response.

Unfortunately, it didn't work. Still unable to communicate with the programmer

[image]

I still don't get it. The hardware VID and PID are correct to me:

[image]

ZakKemble, 2024-01-14:
Are you able to get the command line args that works with platformIO? What version of avrdude are they using? I think this might be more of an issue with avrdude rather than this GUI. https://github.com/avrdudes/avrdude

dzalf, 2024-01-14:
Well, this is getting weird.

Could it be related to different versions being used by the pio environment and the avrdudess GUI?

I located the `tool-avrdude` folder from platformio and ran the same command as in the GUI consode and got what I expected:

dzalf, 2024-01-14:
Yep! My suspicions were right. I believe that there is an issue with the avrdude executable packed with the latest version of this GUI.

I pointed at the executable from the platformio folder and I was able to detect the MCU!

[image]

[image]

ZakKemble, 2024-01-14:
Yea, I would raise this at https://github.com/avrdudes/avrdude since there's nothing I can do here.

dzalf, 2024-01-14:
Got it. Thank you for your support, though.

Best wishes and thank you for the amazing work with avrdudess!

dzalf, 2024-01-14:
Just one last question. Is the AVRDUDESS GUI always compiled from the latest avrdude branch? In other words, you have nothing to do with the executable packed with the GUI, correct?

ZakKemble, 2024-01-14:
Correct, I have nothing to do with avrdude. I just ship the GUI with whatever the latest version of avrdude is at the time.

ZakKemble, 2024-01-14:
Oh I just remembered this issue #62 which might have some clues.

dzalf, 2024-01-14:
> Oh I just remembered this issue #62 which might have some clues.

I did check that issue before, however, I tried with multiple drivers with no luck 😕

mcuee, 2024-01-19:
@dzalf 

You should not use libusb-win32 device driver for Atmel-ICE. You should use native Windows USB HID device driver.

closed 2024-01-26 by ZakKemble