ZakKemble/AVRDUDESS #83: USBasp wrong vendor id

ffw2, 2024-01-14:
Hello,

first, thank you for this great tool.

I have two USBasp programmers from https://protostack.com.au/shop/accessories/usbasp-avr-programmer/
They use the firmware from [fischl.de](https://www.fischl.de/usbasp/) but the vendor id of the programmers is "Protostack Pty Ltd" instead of "fischl.de".
So there is a error message of avrdude.

I reflashed the firmware of the programmers from website of fischl.de, but the vendor id doesn't change.

Change you please add the programmer to AVRDUDESS or add a second vendor id for this programmer type?
Thanks.

[image]

ZakKemble, 2024-01-14:
Hi, there is an entry called "Any usbasp clone with correct VID/PID" with ID "usbasp-clone" which should work.

You'll have to talk to the avrdude guys at https://github.com/avrdudes/avrdude if you want a new entry just for Protostack. h
However I don't think they'll add it since you've already got the clone option above. And they'll just end up with an infinite list of clones to maintain since anyone can change the vendor/product names.

As for the vendor name not changing, that might be fixed by reinstalling the driver, try using Zadig for that https://zadig.akeo.ie/

ffw2, 2024-01-14:
I'm sorry, I overlooked that. Thank you for the information. Now it works!

closed 2024-01-14 by ZakKemble