ZakKemble/AVRDUDESS #88: Windows 11 (won't install)

carl1961, 2024-02-13:
I found version  1.5, 1.6 and this 1.7 installer will not run on windows 11. version installer 1.4 works
I used windows visual studio to compile a release version

By the way Version 2.5 is Great! glad i updated

carl1961, 2024-02-13:
I compiled this 1.7 and made installer and it works 
[AVRDUDESS-2.17-setup.exe.zip](https://github.com/ZakKemble/AVRDUDESS/files/14257978/AVRDUDESS-2.17-setup.exe.zip)
[AVRDUDESS-master.zip](https://github.com/ZakKemble/AVRDUDESS/files/14257981/AVRDUDESS-master.zip)

ZakKemble, 2024-02-13:
Hi, could you provide more info about the installer not working on Windows 11, error messages etc? It seems to be ok for me apart from a few messages about an unknown publisher which is expected.

carl1961, 2024-02-13:
I get no error message, it just does nothing, I will try and see if there is a error log some where.

ZakKemble, 2024-02-18:
Hmm, I've been using Inno Setup 6.2.1 to create the installer. I've now updated to 6.2.2. Could you see if this installer works for you?

[AVRDUDESS-2.17-setup.zip](https://github.com/ZakKemble/AVRDUDESS/files/14324308/AVRDUDESS-2.17-setup.zip)

If that still doesn't work, could you see if anything appears in Event Viewer > Windows Logs > Application (and maybe Setup and System too).

Obiwan28, 2024-04-23:
I have being using an old version of AVRDUDESS in my Windows 10 Pro PC. Excellent program. Today I finished migrating to my new PC with Windows 11 Pro. So I downloaded both versions [AVRDUDESS-2.17-setup.exe] and [AVRDUDESS-2.17-portable.zip] from [here](https://github.com/ZakKemble/AVRDUDESS/releases/tag/v2.17).  It doesn't work for me. It just makes a small blink over the shortcut. As Carl1961 said "_get no error message, it just does nothing_" . I turned my Norton antivirus off, checked for compatibility (does not work). No way and no clue. I will try an older version in the meantime.

Processor	AMD Ryzen 5 5600G with Radeon Graphics            3.90 GHz
Installed RAM	32,0 GB (31,3 GB usable)
Device ID	370CBEC0-A857-4288-A15B-09914CA483C0
System type	64-bit operating system, x64-based processor

Edition	Windows 11 Pro
Version	23H2
Installed on	‎22/‎4/‎2024
OS build	22631.3447
Experience	Windows Feature Experience Pack 1000.22688.1000.0

I have no skills to compile the program by myself, so I will have to wait until a fresh executable version becomes available.
Greetings from Buenos Aires, Argentina.

ZakKemble, 2024-04-24:
I've no idea what's going on there. I installed a fresh Windows 11 23H2 and updated to the exact same OS build etc. as you and AVRDUDESS starts up no problem :/

Could you see if anything appears in Event Viewer > Windows Logs > Application?

Obiwan28, 2024-04-24:
Hello Zak ! I will try to find some trace. My new PC has just 3 days and
now is full of electronics and software development tools, and I am sure
that everything is working reasonably fine. I have no clue. As soon as I
find something I wil be back to you reporting.

Thank you for your concern!
It is just that I like your program so much.

In the meantime, is there a way tp download a previous version? Every time
I find a place offering previous versions such as 1.4 or so, I am
redirected to your page showing the last one.

Greetings from Buenos Aires!

ZakKemble, 2024-04-24:
All versions are here https://github.com/ZakKemble/AVRDUDESS/releases
Click on "Assets" to show download links.

Obiwan28, 2024-04-24:
Dear Zak, 2.14 works perfectly.
2.15 and 2.16 don't. The installer just don't respond.
Very strange. Please, if you find the problem, please let me know.
Thank you !

[image: image.png]

*Roberto G. Berner*
*Professional analogue and digital hardware designer*
*Software developer*
*Industrial control systems & electronic devices*
*Microprocessor theory & applications technical trainer*

*Technical ***@***.***
***@***.***>***@***.***
***@***.***>*
*Cel / WhatsApp: +54 9 11 5472 1234*

Fareeha-Takanashi, 2024-06-20:
Windows Defender disabled The installer. You can try manually allow in Windows Defender it if you want install.
Problem may be in installer maker.
[image]