ZakKemble/AVRDUDESS #92: Malware detected after update to 2.17

pptsk, 2024-04-06:
Hi,
After update my current version of Avrdudess 2.16 to new 2.17 program says there is not avrdude detected and my antivirus Iobit malware fighter says:
avrdude.exe  Gen: Variant.Babar.452770
Also when unpacking portable version 2.17, avrdude is immediatelly deleted by antivirus software.
When instal back version 2.16 everything is fine an program works very well.
What is wrong?
thank you

ZakKemble, 2024-04-06:
Heya, looks like your AV is detecting avrdude v7.3 (not avrdudess) as a false positive. There should be a way to report false positives to the AV vendor.

This is the project page for avrdude - https://github.com/avrdudes/avrdude/

pptsk, 2024-04-06:
Hi,
Thank you!

closed 2024-04-17 by ZakKemble