ZakKemble/AVRDUDESS #97: Clarity of programmer type selection

owenduffy, 2024-06-26:
I note that AVRDUDE has aliases for some programmers... so that in itself creates confusion.

The AVRDUDE programmer type appropriate to the 'original' Arduino bootloader (which implements a cut down set of the STK500 v1 protocol) was called "arduino", so you would specify -c arduino on the ARVDUDE command line.

That choice is no longer available in AVRDUDESS v2.17, seemingly replaced by:

[image]

which does build a command line using -c arduino.

I understand in the confusing environment of multiple programmer types described as arduino or using an arduino that you have provided an expanded label... but most beginners will not be helped by that extra detail.

My suggestion is that it would be clearer that you use the following format:

**arduino (bootloader using STK500 v1 subset)**

So that the literal AVRDUDE argument starts the entry and any explanatory info is in parenthesesis.

Keep up the good work Zak.

Owen

ZakKemble, 2024-06-30:
Yea I agree. Most tutorials and things will be talking about the `-c [id]` argument rather than the full description. I've made it so that it now shows:

`[id] . . . ([description])`
like
`arduino . . . (Arduino for bootloader using STK500 v1 protocol)`

The descriptions are still sourced from the avrdude.conf file and the `. . .` is just to try to break things up a bit otherwise it becomes a massive block of text that's difficult to look through.

owenduffy, 2024-07-03:
Hi Zak, yes sounds good.

A similar thing exists with the -B parameter for USBASP.

If someone has instructions to use -B 10, it cannot be specified directly (AFAIK).

Owen

ZakKemble, 2024-09-07:
Alrighty, added the real -B value in parenthesis for USBasp frequencies. Changes are in v2.18!

closed 2024-09-07 by ZakKemble