ZakKemble/AVRDUDESS #99: Detect atmega328pb

owenduffy, 2024-06-29:
It appears that if I select a MCU = ATmega328PB and click detect, it properly reads the ID as a 328PB, no complaints, but it overwrites the form's MCU to be ATmega328P which of course fails subsequent attempts to read, write etc.

Owen

ZakKemble, 2024-06-30:
I can't reproduce this. I don't have a 328PB to test properly, however faking the returned signature to that of the PB will make it select the correct 328PB part for me.

owenduffy, 2024-07-03:
Apologies... a misunderstanding with my friend who was testing the PB... I was told it was a problem using USBASP (ISP), but in fact that works fine. It is Optiboot that causes the problem, and as you know it fakes the signature (in this case to 328P).

Cheers... Owen.

closed 2024-07-03 by owenduffy