ZakKemble/AVRDUDESS #103: v2.18 doesn't detect ATmega8

ZakKemble, 2024-09-25:
AVRDUDESS-2.18-portable with USBasp (driver libusbK) and AVRDUDE v8.0
doesn't detect ATmega8. With AVRDUDE v7.0 - no problem.

_Originally posted by @forregister84 in https://github.com/ZakKemble/AVRDUDESS/issues/102#issuecomment-2374459426_

ZakKemble, 2024-09-25:
AVRDUDE v8 doesn't output a signature if everything was successful, unlike previous versions.

AmaJuz, 2024-11-06:
Try this

ref : https://electronics.stackexchange.com/questions/416714/avrdude-does-not-recognize-usbasp-device/417509#417509

in conf file

programmer
  id    = "usbasp";
  desc  = "USBasp, http://www.fischl.de/usbasp/";
  type  = "usbasp";
  connection_type = usb;
  usbvid     = 0x16C0; # VOTI
  usbpid     = 0x05DC; # Obdev's free shared PID
  #usbvendor  = "www.fischl.de"; <-- comment out with # as shown
  #usbproduct = "USBasp";  <-- comment out with # as shown

do this for clone as well