ZakKemble/AVRDUDESS #109: Mono refuses to run on macOS

matthewyang204, 2025-03-12:
Mono crashes with this when running without `--arch=32`:
```
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all

=================================================================
	Native Crash Reporting
=================================================================
Got a segv while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

=================================================================
	Native stacktrace:
=================================================================
	0x1063dd679 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_dump_native_crash_info
	0x10637540e - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_handle_native_crash
	0x1063d77f6 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : altstack_handle_and_restore
	0x7ff80bc135e4 - /System/Library/PrivateFrameworks/SkyLight.framework/Versions/A/SkyLight : SLDisplayBounds
	0x10946cec8 - Unknown
	0x1068456b6 - Unknown
	0x1062d2352 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_jit_runtime_invoke
	0x1064e37e2 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_runtime_try_invoke
	0x1064e2408 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_runtime_class_init_full
	0x1062c9d5c - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_jit_compile_method_inner
	0x1062cd8aa - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_jit_compile_method_with_opt
	0x106377f2e - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : common_call_trampoline
	0x106377970 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_magic_trampoline
	0x1067c7393 - Unknown
	0x1090cf41b - Unknown
	0x1092e1a23 - Unknown
	0x1092dda7b - Unknown
	0x10894cbe3 - Unknown
	0x1062d2352 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_jit_runtime_invoke
	0x1064e1e07 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_runtime_invoke_checked
	0x1064ea31e - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : do_exec_main_checked
	0x106332abe - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_jit_exec
	0x106335d25 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : mono_main
	0x1062c1b78 - /Library/Frameworks/Mono.framework/Versions/6.12.0/bin/mono-sgen64 : main
	0x7ff805a5b2cd - /usr/lib/dyld : start

=================================================================
	Telemetry Dumper:
=================================================================
Pkilling 0x123145557663744x from 0x140704348367616x
Entering thread summarizer pause from 0x140704348367616x
Finished thread summarizer pause from 0x140704348367616x.
Failed to create breadcrumb file (null)/crash_hash_0x8a2349e8d

Waiting for dumping threads to resume

=================================================================
	External Debugger Dump:
=================================================================

=================================================================
	Basic Fault Address Reporting
=================================================================
Memory around native instruction pointer (0x7ff80bc135e4):0x7ff80bc135d4  89 fb 48 8b 05 a3 07 da 39 0f 10 00 0f 10 48 10  ..H.....9.....H.
0x7ff80bc135e4  0f 11 4f 10 0f 11 07 89 f7 48 89 de e8 b8 96 f9  ..O......H......
0x7ff80bc135f4  ff 48 89 d8 48 83 c4 08 5b 5d c3 55 48 89 e5 53  .H..H...[].UH..S
0x7ff80bc13604  48 83 ec 28 48 8b 05 51 b5 d8 39 48 8b 00 48 89  H..(H..Q..9H..H.

=================================================================
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at System.Windows.Forms.XplatUICarbon:CGDisplayBounds <0x000b7>
	  at System.Windows.Forms.XplatUICarbon:get_WorkingArea <0x00072>
	  at System.Windows.Forms.XplatUICarbon:get_VirtualScreen <0x00043>
	  at System.Windows.Forms.XplatUI:get_VirtualScreen <0x00048>
	  at System.Windows.Forms.Screen:.cctor <0x0017a>
	  at System.Object:runtime_invoke_void <0x000a5>
	  at <unknown> <0xffffffff>
	  at System.Windows.Forms.Form:get_CreateParams <0x0062a>
	  at DialogForm:get_CreateParams <0x00042>
	  at System.Windows.Forms.Form:SetClientSizeCore <0x001c7>
	  at System.Windows.Forms.Control:set_ClientSize <0x00048>
	  at System.Windows.Forms.Form:set_ClientSize <0x00052>
	  at System.Windows.Forms.Form:set_ClientSize <0x000d2>
	  at System.Windows.Forms.FileDialog:.ctor <0x02662>
	  at System.Windows.Forms.OpenFileDialog:.ctor <0x0003a>
	  at System.Windows.Forms.OpenFileDialog:.ctor <0x0006a>
	  at avrdudess.Form1:InitializeComponent <0x01cc2>
	  at avrdudess.Form1:.ctor <0x000e2>
	  at avrdudess.Form1:.ctor <0x00072>
	  at avrdudess.Program:Main <0x00082>
	  at <Module>:runtime_invoke_void_object <0x000b0>
=================================================================
zsh: abort      mono avrdudess.exe
```

However, when I try the `--arch=32`, it refuses to run. It just says that it failed to switch to mono32.

ZakKemble, 2025-03-12:
Hi, the first error is because mono 64-bit doesn't support WinForms. The second error about mono32 is because Apple removed support for 32-bit applications in macOS Catalina (2019). I should probably update the readme to say that mono doesn't work on modern versions of macOS. However, I seem remember someone saying that they managed to get things working by using some other app from the appstore that costs around $80, but I can't remember what it was called :(

matthewyang204, 2025-03-12:
Yikes, Apple always ruins the fun. First, the removed 32-bit Wine support, forcing us to use SysWOW64 to emulate 32-bit wine apps on 64-bit macOS, reducing the performance of them. And now, it's Mono. So I'd need a Windows or Linux machine to do it? I'm going to be upgrading a 2011 iMac 27" that I got for free to be a cursed Windows workstation anyway in a few weeks, but I was just curious if it would run on macOS. Though you should state that it only works on macOS Mojave and lower in the README, as it may be confusing for some people. Like I initially though Mono used SysWOW64 as well

matthewyang204, 2025-03-13:
I see that the README.md is updated now, closing issue

closed 2025-03-13 by matthewyang204